daqifi/daqifi-nyquist-firmware #3: SCPI ADC Measurement on ADC Ch 8 Causes System Fault

cptkoolbeenz, 2024-03-06:
Channels 0, 1, 2 appear to work.  Channels 3+ appear to crash the mcu.
To reproduce:

SYSTem:POWer:STATe 1
ENAble:VOLTage:DC 3,1
MEASure:VOLTage:DC? 3
Causes crash...

adbenGHM, 2024-03-07:
Upon further investigation I found,
g_BoardData.AInLatest.Size is 3, so comparison against index 3 (line 81, in BoardData.c) fails and BoardData_Get returns NULL. 
Thus pAInLatest (line 46 in SCPIADC.c) in function SCPI_ADCVoltageGet is NULL, so NULL pointer is passed as argument to  ADC_ConvertToVoltage(pAInLatest), which leads to hard fault

cptkoolbeenz, 2024-03-07:
Yes, that makes sense.  I also tested and it works fine.

MAX_AIN_DATA_MOD is used to support multiple ADCs (ADC modules).  So, we can support the onboard ADC, and two external ADCs with MAX_AIN_DATA_MOD = 3.  For the Nq1, I think we should be able to set MAX_AIN_DATA_MOD = 1 since we are only using the internal ADC as defined in NQ1BoardConfig.c ln 36.  I'm not sure how best to reconcile the two - it would be nice to have it defined in one location...

MAX_AIN_CHANNEL is used to store all ADC channel data.  For Nq1, we are using 16 channels for user analog input and an additional 8 channels for internal measurements like rail voltages, temperature, etc.  So, this could be reduced to 24 instead of 48.  Again, it would be nice to have these defined in one location if possible.

closed 2024-12-19 by adbenGHM