daqifi/daqifi-nyquist-firmware #12: Add runtime flag if the device is connected to a client

cptkoolbeenz, 2024-12-10:
Add a runtime flag that indicates the device is connected to a client (USB/WiFi).  This flag should be sent in the sysinfopb SCPI reply such that any client can see if the device is already connected to a client.

adbenGHM, 2024-12-18:
Ok, I will add it

adbenGHM, 2024-12-18:
Under which PB should this info be added ?

cptkoolbeenz, 2024-12-18:
Add this under
`// Device Hardware/Firmware Information`
Maybe `optional uint32 device_con_active = 70`?
Then the uint32 could be a bitwise flag indicating active connection(s) of 0b1=USB, 0b2=TCP, 0b04=UDP, etc.

adbenGHM, 2024-12-18:
Yes, this can be done, 
I was also thinking about using the DaqifiOutMessage_device_status_tag because it is not being used anywhere right now and is 32-bit, or I can rename it with the one you suggested. 
Why do you think?

cptkoolbeenz, 2024-12-18:
Yes, good point.  We created that tag a long time ago for this exact purpose!

adbenGHM, 2024-12-18:
Perfect, i will use it then

adbenGHM, 2024-12-19:
I have added the flags,
Because UDP is not connection oriented, i have added usb, wifi and TCP flags
//01 : USB connected
//02 : wifi connected
//04 : TCP Client Connected

adbenGHM, 2024-12-19:
featured added in  commit c54f96c31a7b198b3bebca5ce2a3871ec4da5ad6

closed 2024-12-19 by adbenGHM