daqifi/daqifi-nyquist-firmware #22: ADC Channel Enable Issue

cptkoolbeenz, 2025-06-07:
Firmware Version: v3.0.0b1
Windows App Version: v3.0.0

@Ryansolar14: Channels 0-4 are working. All other channels show 0V as the input.

Ryansolar14, 2025-06-17:
@cptkoolbeenz: Will this be fixed in a future update?

cptkoolbeenz, 2025-06-19:
Yes, @weehau1984 is working on it.

weehau1984, 2025-06-25:
Multiple test were run to verify this issue.

**Test 1:**
Tested Ain0 to Ain6. All good. However, there are problems for Ain7 to Ain15.
Applied 3.3V to Ain7, but the reading appeared on Ain14. Similarly:
Ain8 showed up as Ain15.
Ain9 showed up as Ain10.
Ain10 showed up as Ain11.
Ain12 showed up as Ain13.
Ain14 showed up as Ain15.
Ain15 - not reading.

**Test 2:**
Re-run the test, but this time all Ain0 to Ain15 were added to the same screen.
All Ain0 to Ain15 appeared to be good.

**Test 3:**
Re-run the test, but only Ain7 to Ain15 were added to the screen.
No signal appeared from Ain7 to Ain15. The issue remained even after closing and reopening the application.

Testing appears to produce varied results. Further testing will be conducted for investigations.

closed 2025-08-04 by cptkoolbeenz